PRBonn/SHINE_mapping #2: bad result for maicity with voxel size 20cm

Rotatingpencil, 2022-12-16:
Thanks for your excellent work!

But when I test the mappings for maicity with voxel size 20 cm, the result turns to like below: 

[image]

It seems there are many holes and the disappear of pavement.

I just simply change the hyperparameter `leaf_vox_size` to 0.2 in `maicity_incre_reg.yaml` . How should I modify it to get a good result?

YuePanEdward, 2022-12-16:
Thanks for your interest in our work.
The issue may be caused by the synthetic dataset and the marching cubes library we used.
You may set the `mc_res_m: 0.1` and you are expected to get a better result without the holes.

Rotatingpencil, 2022-12-20:
Thanks for your reply, I will test it later!

Rotatingpencil, 2023-01-13:
I have tried with `mc_res_m: 0.1`, it has no stranges holes anymore, but the result seems to be less good...
[image]

aartykov, 2023-05-25:
I could not reproduce the results in table 2 with the MaiCit dataset too. What should I do to get similar results as the ones mentioned on the paper? Thanks in advance.

YuePanEdward, 2023-05-26:
> I could not reproduce the results in table 2 with the MaiCit dataset too. What should I do to get similar results as the ones mentioned on the paper? Thanks in advance.

Thanks for your interest in our work. You may check a similar issue here https://github.com/PRBonn/SHINE_mapping/issues/3#issuecomment-1367416983.

closed 2023-06-10 by YuePanEdward