PRBonn/SHINE_mapping #4: How to fix puma's voxel size or feature grid size equal to 10cm?

qixuema, 2023-01-10:
Thank you very much for your outstanding work, and I hope you will be awarded the best paper.
Also, I would like to ask how to ensure the voxel size = 10 cm for the puma method in Table 2 of the paper?

YuePanEdward, 2023-01-10:
Thank you very much for your interest in our work.
Actually, for Puma, we cannot fix the voxel size to a value. Instead, we use the maximum octree depth whose corresponding leaf voxel size is the closest to 10cm. In our experiments,  the maximum octree depth is set as 10 for MaiCity and 11 for the Newer College dataset. Due to space limitations, we didn't mention it in the current paper. We may add it to the updated version.

qixuema, 2023-01-10:
Wow, I get it, thank you so much!

closed 2023-01-10 by qixuema