PRBonn/SHINE_mapping #5: Questions about the time required for KITTI 00 sequence reconstruction.

qixuema, 2023-01-13:
Hello, your work is fantastic!
I would like to ask how long did it take to reconstruct the complete scene of the KITTI 00 sequence?

YuePanEdward, 2023-01-13:
Currently, for the incremental mapping with 0.5m `leaf_vox_size`, the mapping time (without meshing) per frame is about 1s with GPU.  
The timing and memory bottleneck for large-scale reconstruction such as KITTI 00 sequence is actually the mesh reconstruction part. We would predict the SDF value in the whole spatial grid for marching cubes from scratch when doing the meshing and the consumed time would increase continuously. 
You may set the `mesh_freq_frame` to the total frame number to disable the on-the-fly mesh reconstruction and do the offline reconstruction at the end. 
We will later update the more efficient code for grid SDF prediction and marching cubes in chunks to conduct faster mesh reconstruction for large-scale datasets.

qixuema, 2023-01-14:
I see. Indeed, it will take a long time to reconstruct the mesh. I will set `mesh_freq_frame ` to total frame number to reconstruct the KITTI 00 sequence.
Looking forward to your latest version.
Finally, thank you very much for your detailed reply!

closed 2023-01-14 by qixuema

YuePanEdward, 2023-02-02:
We update the code to support a fast and memory-efficient large-scale reconstruction. Now the batch mapping and reconstruction on the whole KITTI 00 sequence costs only about 20 minutes with a marching cube resolution of 15 cm.

qixuema, 2023-02-04:
Wow, that's great, thanks a lot!