PRBonn/SHINE_mapping #6: How can we get the ground truth model?

Rotatingpencil, 2023-01-13:
For MaiCity dataset, I downloaded `gt_map_pc_mai.ply` from `scripts/download_maicity.sh`, I have some questions about this model?

- why it's a pointcloud model rather than a mesh model, is it concatenated from, for example from `mai_city/bin/sequences/00/velodyne/00000.bin` to `mai_city/bin/sequences/00/velodyne/00200.bin` 
- I have seen that  `gt_map_pc_mai.ply` from `scripts/download_maicity.sh` is of sequence 02, so in your paper,  you get the reconstruction quality result on sequence 02? Or it's a average result of sequences 00 to 02?

Looking forward to your reply! Thanks a lot!

YuePanEdward, 2023-01-13:
For MaiCity dataset, sequence 01 and 02 are sampled from the same 3D model (the one inside the `gt_models` folder) with different beam numbers. But sequence 00 is sampled from another 3D model (which is not provided). 

We used sequence 01 for the reconstruction evaluation with the ground truth point cloud generated from the much denser sequence 02. To also take the scene completion quality into account, we combine that point cloud with the ground surface of the original 3D model to generate the `gt_map_pc_mai.ply`. The reason why we do not directly use the original 3D model is that it's not watertight and has multiple layers inside the wall and under the ground (you may open `gt_models/mai_city_block.ply` to check). 

For more details regarding the dataset, you may refer to the original [paper](https://www.ipb.uni-bonn.de/wp-content/papercite-data/pdf/vizzo2021icra.pdf) section **IV A** and also the `README.md` file under the dataset folder.

I hope I have answered your question.

closed 2023-01-13 by YuePanEdward

Rotatingpencil, 2023-01-13:
Thanks a lot, I totally understand.