PRBonn/SHINE_mapping #9: Problem about the input data.

Xiaxia1997, 2023-02-23:
Since I'm going to run SHINE_mapping, on my own dataset, there are several questions :
```
In README.md
Generally speaking,we need to provide:
pc_path : the folder containing the point cloud (.bin, .ply or .pcd format) for each frame.
pose_path : the pose file (.txt) containing the transformation matrix of each frame.
calib_path : the calib file (.txt) containing the static transformation between sensor and body frames (optional, would be identity matrix if set as '').
```
1. Does **the transformation matrix of each frame** means transformation matrix from lidar to world? If we provide it, the **Tr** in calib.txt can be an identity matrix just like the NCD dataset?
2. But for KITTI dataset, the Tr is not an identity matrix. And we applied `poses.append(
            np.matmul(Tr_inv, np.matmul(pose, Tr))
        )  # lidar pose in world frame` in poses.py to get the pose. I tried to understand the meaning of this code.
From [the github issue](https://github.com/PRBonn/semantic-kitti-api/issues/78), we can get
The poses.txt is given in the camera coordinate system, Tr is the extrinsic calibration matrix from velodyne to camera.In this case, 
`Pose_velodyne = Tr_inv * Pose_camera`. But the code is `pose = Tr_inv * Pose_camera *Tr`. I'm confused about it.

Look forward to your favourable reply.

YuePanEdward, 2023-02-23:
Thanks for your interest in our work.
1. If the calib.txt contains an identity matrix, then the pose file should contain the transformation from lidar to world coordinate system.
2. Same as https://github.com/PRBonn/semantic-kitti-api/issues/115.
hope that helps.

closed 2023-02-23 by YuePanEdward

YuePanEdward, 2023-05-17:
Also explained here: https://github.com/PRBonn/kiss-icp/issues/156#issuecomment-1551695713