PRBonn/SHINE_mapping #12: why are there some unusual extra artifacts in the reconstruction result.

marrblue, 2023-03-16:
Great job I must to say first.
Here I just found some strange results in my tests as shown in the screen shot.
I used shine_batch.py to rebuild maicity model, leaf voxel size, mc_res was set to 0.2m, 0.1m respectively. iters = 20000.
Other parameters keep defaults basically.
It seems like there's an extra part of the final result. I am wondering if you have any clues and ideas for this.
[image]
[image]
The distance between the "false wall" and the real wall is about 30cm

StarryN, 2023-03-16:
Hello @marrblue, Thanks for your interest in our work.
The reason is that we have no real observation behind the wall, and under the current mechanism, the not observed area will be regarded as free (SDF > 0) by default. But behind the wall, a small area is labeled as occupied (SDF < 0) because of our sampling strategy, then the 0 level-set boundaries appeared.
In the old version code, we calculate SDF uniformly in the whole space but only extract mesh from the octree leaf node. In this way, we can avoid extracting most of the fake wall. And in the current version, we only calculate SDF in the leaf nodes to remove even more fake walls than the old one.
I hope I have answered your question.

marrblue, 2023-03-16:
Hi @StarryN !  your answer is very reasonable and I will try to avoid this problem by modifying related codes. Thank you!

closed 2023-03-16 by StarryN