PRBonn/SHINE_mapping #18: semantic loss

cv-lab-x, 2023-06-07:
hi, thanks for your great work, you mentioned semantic loss in your tips and codes, but there're no ablation study in your paper,  did you do the ablation study about the semantic loss ?    
Looking forward to your reply, thanks!
@niosus @stachnis @nivedchebrolu @ovysotska @StarryN

niosus, 2023-06-07:
@cv-lab-x please do not tag everyone without a second thought. Most people you tagged are not involved in this paper.

StarryN, 2023-06-08:
Hi @cv-lab-x, Thanks for your interest in our work, and you don't need to @ so many people to get response in our lab's projects :-). 
The semantic loss is just used to show our method can be easily extended to semantic 3D reconstruction (output semantic label for each query point) with only tiny modifications, like adding semantic loss with another semantic MLP decoder. We didn't test whether the semantic loss can improve our map quality, but from my point of view, it won't make a significant difference.

closed 2023-06-10 by YuePanEdward