PRBonn/SHINE_mapping #22: each leaf node store multiple points?

ZZy129326999, 2023-07-12:
Hello, may I ask, can each leaf node store multiple points?

StarryN, 2023-07-12:
Hello @ZZy129326999, We store one-dimensional feature vectors rather than raw points in our grids.  And the points serve as training data to supervise the optimization of our map.

ZZy129326999, 2023-07-13:
> Hello @ZZy129326999, We store one-dimensional feature vectors rather than raw points in our grids. The input points serve as training data to supervise the optimization of our map and they are not stored in any special data structure.

Thanks for your answer. I still have two questions. The first question is whether the incremental construction of octree is reflected in the normalized point box of each frame. Once new raw points appear, a new node will be created under level-0 of octree , but if the point is not deleted in this way, will the gpu video memory keep increasing? Furthermore, will new normalized points appear, will they overwrite the original points, or will the shift point handle this problem, without explicitly updating the octree, and offset the point to simulate the incremental process? The second question is to obtain the corresponding node through the 3D coordinates of the raw points, followed by the morton code, and then the center. How to understand the center and node?

ZZy129326999, 2023-07-13:
Oh, I understand, it is the sliding window mode, thank you!