PRBonn/SHINE_mapping #23: Weird about the calculation of free space uniform sampling ?

DeepDuke, 2023-07-17:
```cpp
       # Part 3. free space uniform sampling
        repeated_dist = distances.repeat(freespace_sample_n,1)
        free_max_ratio = free_sample_end_dist_m_scaled / repeated_dist + 1.0
        free_diff_ratio = free_max_ratio - free_min_ratio

        free_sample_dist_ratio = torch.rand(point_num*freespace_sample_n, 1, device=dev)*free_diff_ratio + free_min_ratio
        
        free_sample_displacement = (free_sample_dist_ratio - 1.0) * repeated_dist
```
About the above caclualtion in [https://github.com/PRBonn/SHINE_mapping/blob/master/utils/data_sampler.py](url), my understanding is samping points from 0.3m to 0.8m as the distance to the surface (for `sigma=0.1 m`, then surface sampling range should be `[-0.3m, 0.3m]` around the surface). But the above calculation is a bit weird to me

In most of the config files, we have  `free_sample_begin_ratio: 0.3` and ` free_sample_end_dist_m: 0.8`. Suppose there is a scan point with range to the LiDAR as `repeated_dist = 10m`. Then `free_max_ratio = 0.8 / 10 + 1.0 = 1.08`, `free_diff_ratio=1.08 - 0.3=0.78`, then    `0.3 <= free_sample_dist_ratio <= 1.08`, finally `-7m < free_sample_displacement < 0.8m`.  This means a high probability to sample a point from opposite direction with respect to the orignal scan point.  Is it the meaning of free samping ? Or where is wrong with my understanding? Thank you!

YuePanEdward, 2023-07-19:
Thanks a lot for your interest in our work. 
In my implementation, we have two types of samples: 
1. the close-to-surface samples (indicated by the parameter `surface_sample_range_m`, which is sampled in the range `[-surface_sample_range_m, surface_sample_range_m]` along the ray from the endpoint)
2. the free space samples (indicated by the parameters `free_sample_begin_ratio` and `free_sample_end_dist_m`, which is sampled in the range `[-free_sample_begin_ratio * ray_distance, free_sample_end_dist_m]` along the ray from the endpoint).

Note that the sign is opposite from the SDF label (`+`: behind the surface, `-`: in front of the surface) here.
This implementation is just for convenience since the free space samples may also lie in the close-to-surface region according to the current implementation. You may also split the free space into two parts ( `[-free_sample_begin_ratio * ray_distance, -surface_sample_range_m]` and `[surface_sample_range_m, free_sample_end_dist_m]` ) but I think the results will not have a big difference.

DeepDuke, 2023-07-20:
@YuePanEdward Thanks for your kind explanation.  I previously thought `+` is in front of the surface...

closed 2023-07-20 by DeepDuke