PRBonn/SHINE_mapping #24: Colored Mesh

DimitrisKatsatos, 2023-07-26:
Hi @YuePanEdward,

Thanks for your great work. I have already tested out Kiss-ICP several times, and now I experimented with SHINE, which seems to work better for 3D reconstruction of outdoor scenes. May I ask, if there is a way to generate a colored MC mesh? I'm using point clouds from a stereo cam, not lidar.

King regards

YuePanEdward, 2023-07-26:
Hi, @DimitrisKatsatos, thanks for your interest in our work.
It's possible to generate a colored mesh but it's currently not implemented in SHINE-Mapping.
One possible solution is to use another MLP decoder to regress the color value using the same interpolated feature vector (similar to what we did for the semantic mapping).
You are welcome to send a PR to add the feature of colorized mesh reconstruction to SHINE-Mapping.

ZorAttC, 2023-12-26:
@YuePanEdward @ddkats  Hi,I have implemented the colored mesh method based on the SHINE-MAPPING repository.Hope it can help you. [click this](https://github.com/ZorAttC/COLOR_SHINE_MAPPING.git)

YuePanEdward, 2024-01-05:
@ZorAttC Thank you very much for your contribution. I will mark this issue as closed.

closed 2024-01-05 by YuePanEdward