PRBonn/SHINE_mapping #27: Question on the sdf map

ljjTYJR, 2023-08-15:
Hi all, thanks for the excellent work.

Here I have a minor question on the generated SDF map.

If I understands correctly, in the default configuration, the octree tree is only constructed near the surface and maintained by a `dict` structure.

For those "free space" and "unexplored area" which do not lie in the constructed octree, their feautres will be allocated by `0.`, the generated SDF value may be random during optimization.

Did I understand correctly?

Best,
Shuo

StarryN, 2023-08-15:
Hello @ljjTYJR , Thanks for your interest.
Yes, you are right. The free space and unobserved area will be assigned all zero feature vectors. So the sdf predictions in such areas are all the same and depend entirely on the MLP's bias. We also sample some points in the free space and give them labels near 0 (after scaled sigmoid). So, the prediction of all zero feature vectors will converge to a positive value larger than the truncation distance.

ljjTYJR, 2023-08-15:
@StarryN 
Thanks for your reply.

In fact, one thing always confuses me is that how can we use the neural implict representation in the robotics application.
For the downstream application such as localization, path planning and so on.

When we can reconstruct a denser map from the point cloud, but how to use it in practice?
Do you have any idea?

StarryN, 2023-08-16:
Hello @ljjTYJR.  
For localization, you may check works from our lab like [LocNDF](https://github.com/PRBonn/LocNDF) and [ir-mcl](https://github.com/PRBonn/ir-mcl). They showed that the neural map representation can be used for scan-to-map registration and Monte Carlo localization.
For path planning of mobile robots, In my opinion, The neural SDF field can provide continuous and maybe more accurate gradient vectors which can be used in trajectory optimization, like using ESDF map in [Fast-Planner](https://github.com/HKUST-Aerial-Robotics/Fast-Planner)

closed 2023-08-17 by ljjTYJR