PRBonn/SHINE_mapping #28: some question about the metric accuracy?

Zhangjyhhh, 2023-09-05:
Hi, thanks for your excellent work?
I run the evaluator.py on maicity dataset . I get the similar result as your result,except accuracy. my accuracy is 3.3 that corresponds to "MAE_accuracy". I'm not sure that is  "MAE_accuracy"  correspond to the accuracy?

ljjTYJR, 2023-09-05:
I guess the issue comes from the selection of ground truth.
Refer to https://github.com/PRBonn/SHINE_mapping/issues/3#issuecomment-1367416983, which uses "Interaction point cloud" when evaluating accuracy

ljjTYJR, 2023-09-06:
But I have a question that I used the provided configure file, I can not reproduce the same result as perfect as the result in https://github.com/PRBonn/SHINE_mapping/issues/3#issuecomment-1367416983

Zhangjyhhh, 2023-09-06:
I use the "gt_map_pc_mai_inter_croped.ply" as ground truth, except "MAE_accuracy",other metric is better than the paper's metric.maybe you can use the /config/configicra2023  @ljjTYJR . 
so the MAE_accuracy is the accuracy ?

ljjTYJR, 2023-09-06:
This is the reconstruction in https://github.com/PRBonn/SHINE_mapping/issues/3#issuecomment-1367416983
[image]

This is the reconstruction I use the https://github.com/PRBonn/SHINE_mapping/blob/master/config/config_icra2023/maicity_batch.yaml
[image]
In fact, I think the reconstruction with the above configuration file is already good. But I wonder how can I get the first one? It is really nice!

Zhangjyhhh, 2023-09-07:
I have no idea， maybe you can ask the author, I wish you can figure out

closed 2023-12-11 by YuePanEdward