PRBonn/SHINE_mapping #33: Cuda - torch - kaolin dependencies

HamzaOuajhain, 2024-01-13:
Hello, 

I am trying to run this project, I have an Ubuntu 22.04 with cuda V11.8

nvcc --version returns V11.8.89

nvidia-smi Driver Version: 545.23.08    CUDA Version: 12.3

I did run the recommanded install instruction by pytorch 
conda install pytorch torchvision torchaudio pytorch-cuda=11.8 -c pytorch -c nvidia

and got torch version 1.13.1

the problem is that when I see the compatibility of kaolin, I require installing torch-2.0.0 or 2.1.1

[image]

How can I procced to solve this problem ? should I downgrade cuda ?

HamzaOuajhain, 2024-01-14:
My appologies for my ignorance. I managed to change the paths inside the the _batch.yaml files and get it to work.

  pc_path: "./data/mai_city/ply/sequences/01/velodyne"
  pose_path: "./data/mai_city/ply/sequences/01/poses.txt"
  calib_path: "./data/mai_city/ply/sequences/01/calib.txt"

I will take this change to ask a question regarding running of the program most of the time the visualiser crashes

[image]

when running python shine_incre.py ./config/maicity/maicity_incre_replay.yaml

is this caused by my graphic card or cpu, my cpu is 100%

any help would be appreciated

StarryN, 2024-01-14:
Hello, @HamzaOuajhain. Thanks for your interest.
The warning message you showed means that the open3d visualizer is waiting too long for the next update. You may just click "wait" and wait for the completion of the next processing loop.

HamzaOuajhain, 2024-01-15:
@StarryN Got it, Thank you

HamzaOuajhain, 2024-01-15:
@StarryN  Got it, Thank you.

closed 2024-01-15 by HamzaOuajhain