PRBonn/SHINE_mapping #37: issue

whwh747, 2024-09-07:
Thank you for your excellent work. I used kiss icp + python shine_batch.py config/maicity/maicity_batch.yaml and got some strange results. What could be the reason for this?
[image]

YuePanEdward, 2024-09-08:
Hi, I guess the reason is that the pose estimated by KISS-ICP on this sequence using the default parameter is not correct.

[image]

As I also mentioned in this issue https://github.com/PRBonn/PIN_SLAM/issues/28, the simulation of the poses in MaiCity does not obey the common kinematics of robots. You may check here for more details: https://github.com/PRBonn/kiss-icp/issues/62. Mai-city was not designed to benchmark odometry systems.

whwh747, 2024-09-09:
thank you，I see, perhaps we can provide the poses obtained from another odometry to SHINE.

closed 2024-09-09 by whwh747