PRBonn/SHINE_mapping #40: Waymo Dataset

YuangJia, 2025-02-23:
First of all, thanks for such a great work! When I try to build a map on the Waymo dataset, the downsampled point cloud is normal, but the final reconstructed mesh is very sparse. I still don't understand why this is. Can you give me some hints? I tested it on the sequence 10017090168044687777_6380_000_6400_000

[image]

[image]

YuangJia, 2025-02-23:
By the way, my config settings are as follows:

setting:
  name: "waymo_batch"
  output_root: "./experiments/"
  dataset: "waymo"
  scene_path: "/data0/jiayuang/TSDF-GS/data/waymo/10017090168044687777_6380_000_6400_000.txt"
  label_path: "/data3/jiayuang/SHINE_mapping-master/data/kitti_example/sequences/00/labels" # optional for semantic mapping
  pc_path: "/nvme0/jiayuang/TSDF-GS-2/data/waymo/10017090168044687777_6380_000_6400_000"
  pose_path: "/nvme0/jiayuang/TSDF-GS-2/data/waymo/10017090168044687777_6380_000_6400_000.txt"
  calib_path: "/nvme0/jiayuang/TSDF-GS-2/data/waymo/calib.txt" # kitti format
  load_model: False
  model_path: ""
  first_frame_ref: True
  begin_frame: 0
  end_frame: 5000
  every_frame: 1 # 1 means does not skip
  device: "cuda"
  gpu_id: "0"
process:
  min_range_m: 3.0
  pc_radius_m: 50.0  # distance filter for each frame
  min_z_m: -4.5
  rand_downsample: False # use random or voxel downsampling
  vox_down_m: 0.5
  rand_down_r: 0.2
sampler:
  surface_sample_range_m: 0.3
  surface_sample_n: 3
  free_sample_begin_ratio: 0.3
  free_sample_end_dist_m: 0.8
  free_sample_n: 3
octree:
  leaf_vox_size: 0.2
  tree_level_world: 13
  tree_level_feat: 3
  feature_dim: 8
  poly_int_on: True
  octree_from_surface_samples: True
rgb:
  rgb_on: False
  rgb_mlp_hidden_dim: 32
  rgb_mlp_level: 2
  rgb_mlp_bias_on: True
  weight_rgb: 0.2
decoder:
  mlp_level: 2
  mlp_hidden_dim: 32
  freeze_after_frame: 0
loss:
  ray_loss: False
  main_loss_type: sdf_bce  # select from sdf_bce (our proposed), sdf_l1, sdf_l2, dr, dr_neus
  sigma_sigmoid_m: 0.1
  loss_weight_on: False
  behind_dropoff_on: False
  ekional_loss_on: True
  weight_e: 0.1
continual:
  continual_learning_reg: False # using incremental mapping with regularization
  lambda_forget: 0 # the larger this value, the model would be less likely to forget
  window_replay_on: False # replay within the sliding window
  window_radius_m: 0
optimizer:
  iters: 40000
  batch_size: 4096
  learning_rate: 0.05
  weight_decay: 1e-7 # l2 regularization
eval:
  wandb_vis_on: False # log to wandb or not
  o3d_vis_on: False # visualize the mapping or not
  vis_freq_iters: 10000 # reconstruct the mesh every x iterations
  save_freq_iters: 10000 # save the model and octree every x iterations
  mesh_freq_frame: 1
  mc_res_m: 0.2 # reconstruction marching cubes resolution
  mc_with_octree: False # querying sdf only in certain levels of the octree
  mc_local: False
  mc_vis_level: 3
  save_map: False # save the sdf map or not

YuePanEdward, 2025-02-24:
Thanks for your interest in our work.
You may first check whether the input point cloud is in the coordinate system of the Top LiDAR.
I encountered similar issues previously when the input merged point cloud (from 5 LiDARs) was in the car body frame, whose z value is very close to the ground. In such a case, the raycasting condition would not be fulfilled, thus leading to unexpected reconstruction results.

YuangJia, 2025-02-24:
Thanks for your reply! I think you are right. I used mmdetection3d to convert the waymo dataset to kitti format. The point clouds of the five lidars are fused together and are in the ego-vehicle coordinate system. At present, I have found the conversion matrix from the top lidar to the ego-vehicle coordinate system and aligned it to the world coordinate system, but the reconstruction effect is still not ideal. Do you have any good suggestions?

[image]

YuangJia, 2025-02-24:
Hi, I have fixed this problem. I modified some codes of mmdetection3d, only extracted the point cloud of top_lidar and converted it to lidar coordinate system, and then calculated the matrix of lidar2world.

[image]

closed 2025-02-24 by YuangJia